arpg-sophisticated/ar-tu-do #3: VMWare VM aufsetzen

MaLangenk, 2019-10-30:


marian42, 2019-10-30:
Ich weiß nicht genau, was ihr vor habt, aber wir haben letzes Jahr auch versucht, mit VMs zu arbeiten und haben das nicht hinbekommen. Am Ende hat jeder von uns sich einfach Ubuntu installiert. cc @stefanjva

StefanJVA, 2019-10-30:
> Ich weiß nicht genau, was ihr vor habt, aber wir haben letzes Jahr auch versucht, mit VMs zu arbeiten und haben das nicht hinbekommen. Am Ende hat jeder von uns sich einfach Ubuntu installiert. cc @StefanJVA

Yep das ist in der Tat problematisch. Insbesondere Gazebo will einfach nicht gut funktionieren in einer VM. Hat wohl teilweise damit zu tun, dass man innerhalb einer VM meist nur sehr alte GL Versionen hat. Wir haben auch viel mit Docker rumprobiert, aber sind daran leider auch gescheitert. Das einzige, das halbwegs funktioniert hat ist Gazebo auf dem Linux-Subsystem von Windows 10 zum laufen zu bekommen.

marcelebbrecht, 2020-04-08:
kann weg, ist kunst

closed 2020-04-08 by marcelebbrecht