arpg-sophisticated/ar-tu-do #4: Simulation only starts sometimes

marcelebbrecht, 2019-11-08:
I need 4-8 attempts to get the simulation run properly. Cannot see any pattern.... 

Questions:
1. How many attempts do others need
2. Sometimes I got an error regarding a filled queue. For sure we got dependencies between the processes. Do you also got those messages? 
3. Who wants to fiddle this?

We need a reliable starting behavior!!!
My ideas:
Check deps of nodes/processes
Start dependent process when dependencies are ready
Or simple hack: If possible add sleep between starting the single processes / nodes

Maybe this ist just a simple timing problem caused by fully loaded CPUs....

marian42, 2019-11-10:
This is most probably an issue with gazebo. On my machine, Gazebo starts successfully more often than it doesn't.

marcelebbrecht, 2019-11-10:
Same Code on different machines behaves different... Points to Hardware or
OS.... or Timing. Lets compare the startup process on Wednesday.

marcelebbrecht, 2019-11-13:
Mal eine übliche Ausgabe. Asuwirkung: Alle Fenster da, ich sehe nur keine Strecke im Gazebo und kein Lidar im RViz

Gazebo multi-robot simulator, version 7.16.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 7.16.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[INFO] [1573634491.049698, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ERROR] [1573634491.326012881]: Couldn't open joystick /dev/input/js0. Will retry every second.
SpawnModel script started
[ INFO] [1573634491.395542271]: Finished loading Gazebo ROS API Plugin.
[Msg] Waiting for master.
[ WARN] [1573634491.405192690]: updateConfig() called on a dynamic_reconfigure::Server that provides its own mutex. This can lead to deadlocks if updateConfig() is called during an update. Providing a mutex to the constructor is highly recommended in this case. Please forward this message to the node author.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 129.217.73.190
[ INFO] [1573634491.411270156]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1573634491.421420111]: Finished loading Gazebo ROS API Plugin.
[Msg] Waiting for master.
[ INFO] [1573634491.423067476]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 129.217.73.190
[INFO] [1573634491.522806, 0.000000]: Loading model XML from ros parameter
[INFO] [1573634491.526769, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
libGL error: failed to create drawable
libGL error: failed to create drawable
[INFO] [1573634493.044271, 0.000000]: Calling service /gazebo/spawn_urdf_model
[ INFO] [1573634493.695172151, 0.001000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1573634493.699678107, 0.001000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1573634493.703708968, 0.001000000]: <robotNamespace> set to: //
[ INFO] [1573634493.703764477, 0.001000000]: <topicName> set to: //imu
[ INFO] [1573634493.703785026, 0.001000000]: <frameName> set to: imu_link
[ INFO] [1573634493.703818667, 0.001000000]: <updateRateHZ> set to: 100
[ INFO] [1573634493.703843884, 0.001000000]: <gaussianNoise> set to: 0
[ INFO] [1573634493.703875798, 0.001000000]: <xyzOffset> set to: 0 0 0
[ INFO] [1573634493.703930123, 0.001000000]: <rpyOffset> set to: 0 -0 0
[Wrn] [msgs.cc:1808] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[gpu_ray] not suppported.
[INFO] [1573634493.723559, 0.001000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1573634493.787753831, 0.001000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1573634493.787868850, 0.001000000]: Starting GazeboRosLaser Plugin (ns = /)
[ INFO] [1573634493.794017840, 0.001000000]: GPU Laser Plugin (ns = /) <tf_prefix_>, set to ""
[ INFO] [1573634493.795528108, 0.001000000]: LoadThread function completed
[ INFO] [1573634493.854462893, 0.001000000]: Loading gazebo_ros_control plugin
[ERROR] [1573634493.854646865, 0.001000000]: GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.

[ INFO] [1573634493.854742281, 0.001000000]: Starting gazebo_ros_control plugin in namespace: /racer
[ INFO] [1573634493.855993565, 0.001000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[racer_spawn-4] process has finished cleanly
log file: /home/marcel/.ros/log/62e3f27a-05f1-11ea-95a3-f8633f28dbf1/racer_spawn-4*.log
[ INFO] [1573634494.015514109, 0.001000000]: Loaded gazebo_ros_control.
[ INFO] [1573634494.044391675, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1573634494.045747005, 0.024000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1573634494.111853, 0.040000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1573634494.114203, 0.041000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1573634494.117619, 0.042000]: Loading controller: left_wheel_back_velocity_controller
[Wrn] [msgs.cc:1808] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[gpu_ray] not suppported.
[ INFO] [1573634494.231549349, 0.078000000]: Physics dynamic reconfigure ready.
[ INFO] [1573634494.278901459, 0.094000000]: Physics dynamic reconfigure ready.
[INFO] [1573634494.377888, 0.130000]: Loading controller: right_wheel_back_velocity_controller
[INFO] [1573634494.522218, 0.183000]: Loading controller: left_wheel_front_velocity_controller
[INFO] [1573634494.602052, 0.202000]: Loading controller: right_wheel_front_velocity_controller
[INFO] [1573634494.737457, 0.239000]: Loading controller: left_steering_hinge_position_controller
[INFO] [1573634494.823435, 0.275000]: Loading controller: right_steering_hinge_position_controller
[... cut ...] ition_controller, joint_state_controller
[INFO] [1573634494.962428, 0.363000]: Started controllers: left_wheel_back_velocity_controller, right_wheel_back_velocity_controller, left_wheel_front_velocity_controller, right_wheel_front_velocity_controller, left_steering_hinge_position_controller, right_steering_hinge_position_controller, joint_state_controller
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/racetrack/pose/local/info, deleting message. This warning is printed only once.

marcelebbrecht, 2019-11-13:
Was gut hilft: gui:=false beim Start setzen, dann wird der Gazebo Client nicht gestartet. Erfolgsquote geht dann gegen 100%.

closed 2020-05-14 by marcelebbrecht