arpg-sophisticated/ar-tu-do #13: Auslesen Kamera (Auto)

timogojo, 2019-11-13:
Unter ROS sollen die Werte der Stereokamera abgerufen und ausgegeben werden

marcelebbrecht, 2020-01-19:
Über Kinect Modul in Gazebo eingebaut, bisher nur darstellung ohne Funktion, nächster Step ist die Umsetzung am Auto.

marcelebbrecht, 2020-01-27:
cam is included on car rviz, will be testet on next wednesday. 

If successfull, we'll need a filter to create a laserscan like line for comparision.

marcelebbrecht, 2020-01-27:
2. Laserscan wird jetzt aus kamera erzeugt

marcelebbrecht, 2020-01-27:
will be tested on car, the finished

marcelebbrecht, 2020-04-04:
Update of SDK leaded to more accuracy for simulated laserscan. Request merge at this point and close.

marcelebbrecht, 2020-04-04:
Next step would be a SLAM approach based on zed mapping feature in a different branch. Therefore we need a simulation of fused pointcloud with  rc_cloud_accumulator in gazebo.

closed 2020-04-10 by marcelebbrecht