arpg-sophisticated/ar-tu-do #20: Aufsetzen Server

timogojo, 2019-11-13:
Es soll ein Server der eine Funktionierende Entwicklungsumgebung / Simulationsumgebung enthält aufgesetzt werden.

marcelebbrecht, 2019-11-13:
Server soweit installiert, geh morgen vermutlich ans Netz. Ich mache schon mal die Doku, wenn der online ist, eben ein paar nacharbeiten und gebe Bescheid, wenn der fertig und benutzbar ist.

marcelebbrecht, 2019-11-14:
Dauert leide rnoch eine Woche, haben jetzt ne IP in der DMZ bzw. wird bis nächste Woche alles vom IRB vorbereitet. System bekommt, wenn es online ist, den letzten Schliff

marcelebbrecht, 2019-11-14:
skripte geschrieben, bitte in die toad.example schauen, erklärt sich aber beim starten der toad.sh von selbst

marcelebbrecht, 2019-12-18:
läuft soweit

marcelebbrecht, 2020-01-27:
wird demnächst nochmal auf eine andere Maschine geschoben, kann aber zu!

closed 2020-01-27 by marcelebbrecht