arpg-sophisticated/ar-tu-do #131: Stable in Gazebo

marcelebbrecht, 2020-08-12:
WF2 and 5 fails in simulation, please ensure, that both, or at least wf5 can run this track.

jhbruhn, 2020-08-12:
Fails to start or due to collisions?

marcelebbrecht, 2020-08-12:
crash bandicot style ....

even on low speed, the second 90° is deadly ... just run gazebo from
current wallfollowing-improvements branch, you'll see

jhbruhn, 2020-08-14:
First improvements can be seen by setting the weight of the car in the simulation to 9kg (instead of 3.5kg). What value is the real value here? I have 9kg in my head but I don't know if that's the actual weight of the car. 
Otherwise, at least for WF5, this is a thing of properly tuning the parameters for the track. I have done some testing and got nearer to a good driving performance by changing parameters, but this only makes sense as soon as the algorithm is "finished" -> parameter impacts don't change anymore.
About WF2: I am unsure about how well parameter tuning works here, still have to try.

Rothu, 2020-08-16:
WF2 doesn't run properly on wallfollowing-improvements anymore, because I added a speed correction factor in car_controller to get the intern speed value closer to reality.

jhbruhn, 2020-08-17:
Can confirm, adding that factor to wallfollowing2 indeed fixes the simulation of it.